Andy1978/hf2gcode #2: hf2gcode interactive demo

koppi, 2016-04-22:
Hi @Andy1978 . I've created a small GUI to preview G-Code in a more interactive manner, see:

https://github.com/koppi/gcoder and Video: https://www.youtube.com/watch?v=9D3hMXP5-QM

One thing in ```hf2gcode.c``` that needs to be changed imho is, to send the gcode data to stdout and the debug and info messages to stderr. Should I send you a pull request to make this happen?

Andy1978, 2016-04-23:
Hi @koppi . This looks very cool, thank you for sharing. Which part in hf2gcode.c so you want to change? I think you want something like "hf2gcode -q -m "hello world"" which only has the gcodes and nothing else. Or what do you want to change? Feel free to send a pull request. Happy hacking, Andy

koppi, 2016-04-23:
Fixed in https://github.com/Andy1978/hf2gcode/commit/252d78a0e1c5ca4041f9e1a3fd6db9a48bf0f38e .

closed 2016-04-23 by koppi

Andy1978, 2016-04-23:
And for the record: You increased the BUFSIZE, this shouldn't be needed. It's used to read BUFSIZE chunks from the file and reallocates memory if it doesn't fit into memory. The reason you saw a problem is fixed in https://github.com/Andy1978/hf2gcode/commit/c75ab47f9912d9fb33c91aa947e39b30917ff8a1