Andy1978/hf2gcode #6: G-Code output too long for GRBL and LinuxCNC

koppi, 2016-05-16:
Please split the text and add some line-breaks to the "( text='...' )" G-Code output:

https://github.com/Andy1978/hf2gcode/blob/master/src/hf2gcode.c#L298

hf2gcode generates G-Code with long input texts does not work on GRBL and LinuxCNC:
 
* The GRBL motion controller has a line length limit (– 50 for older versions, 70 for 0.8c dev or later),
* The LinuxCNC / Machinekit G-Code interpreter has a line length limit, too (– don't know the exact values but just ran into this):

```
Interpreter exited with error:
Command too long
( text="123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456
```

Andy1978, 2016-05-19:
Hi koppi,
the line you pasted is just a comment. I can't check it right now but does GRBL and LinuxCNC really fail if a comment (which they just should ignore) is longer than some preset limit? Of course you can call hf2gcode with "-m" (Generate minimalistic g-code, suppress comments) to suppress any comments which should solve your problem.

koppi, 2016-05-19:
Both GRBL and the NIST interpreter fail if you send them this long comment. I still like to have all these comments in the G-Code file, so the -m option is a work-around that does however not really solve the problem.

koppi, 2016-06-05:
I just want to keep the text in the comments as info.

closed 2019-10-19 by koppi