Andy1978/hf2gcode #8: Didn't get proper gcode file

seesunmoon, 2019-10-24:
Hi there,

**I tried to build this project:**

> make -C ../hershey_fonts src
make[1]: Entering directory '/home/pi/dev/hf2gcode/hershey_fonts'
./fix_linebreak.sh orig/greeks.jhf fixed/greeks.jhf
removing unwanted linebreak and CR in orig/greeks.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/greeks.jhf","gen_c_src/greeks.h")'
./fix_linebreak.sh orig/gothiceng.jhf fixed/gothiceng.jhf
removing unwanted linebreak and CR in orig/gothiceng.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/gothiceng.jhf","gen_c_src/gothiceng.h")'
./fix_linebreak.sh orig/timesr.jhf fixed/timesr.jhf
removing unwanted linebreak and CR in orig/timesr.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/timesr.jhf","gen_c_src/timesr.h")'
./fix_linebreak.sh orig/music.jhf fixed/music.jhf
removing unwanted linebreak and CR in orig/music.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/music.jhf","gen_c_src/music.h")'
./fix_linebreak.sh orig/astrology.jhf fixed/astrology.jhf
removing unwanted linebreak and CR in orig/astrology.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/astrology.jhf","gen_c_src/astrology.h")'
./fix_linebreak.sh orig/scripts.jhf fixed/scripts.jhf
removing unwanted linebreak and CR in orig/scripts.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/scripts.jhf","gen_c_src/scripts.h")'
./fix_linebreak.sh orig/cyrilc_1.jhf fixed/cyrilc_1.jhf
removing unwanted linebreak and CR in orig/cyrilc_1.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/cyrilc_1.jhf","gen_c_src/cyrilc_1.h")'
./fix_linebreak.sh orig/gothgbt.jhf fixed/gothgbt.jhf
removing unwanted linebreak and CR in orig/gothgbt.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/gothgbt.jhf","gen_c_src/gothgbt.h")'
./fix_linebreak.sh orig/symbolic.jhf fixed/symbolic.jhf
removing unwanted linebreak and CR in orig/symbolic.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/symbolic.jhf","gen_c_src/symbolic.h")'
./fix_linebreak.sh orig/futuram.jhf fixed/futuram.jhf
removing unwanted linebreak and CR in orig/futuram.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/futuram.jhf","gen_c_src/futuram.h")'
./fix_linebreak.sh orig/cyrillic.jhf fixed/cyrillic.jhf
removing unwanted linebreak and CR in orig/cyrillic.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/cyrillic.jhf","gen_c_src/cyrillic.h")'
./fix_linebreak.sh orig/timesib.jhf fixed/timesib.jhf
removing unwanted linebreak and CR in orig/timesib.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/timesib.jhf","gen_c_src/timesib.h")'
./fix_linebreak.sh orig/mathlow.jhf fixed/mathlow.jhf
removing unwanted linebreak and CR in orig/mathlow.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/mathlow.jhf","gen_c_src/mathlow.h")'
./fix_linebreak.sh orig/greek.jhf fixed/greek.jhf
removing unwanted linebreak and CR in orig/greek.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/greek.jhf","gen_c_src/greek.h")'
./fix_linebreak.sh orig/markers.jhf fixed/markers.jhf
removing unwanted linebreak and CR in orig/markers.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/markers.jhf","gen_c_src/markers.h")'
./fix_linebreak.sh orig/timesg.jhf fixed/timesg.jhf
removing unwanted linebreak and CR in orig/timesg.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/timesg.jhf","gen_c_src/timesg.h")'
./fix_linebreak.sh orig/meteorology.jhf fixed/meteorology.jhf
removing unwanted linebreak and CR in orig/meteorology.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/meteorology.jhf","gen_c_src/meteorology.h")'
./fix_linebreak.sh orig/japanese.jhf fixed/japanese.jhf
removing unwanted linebreak and CR in orig/japanese.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/japanese.jhf","gen_c_src/japanese.h")'
./fix_linebreak.sh orig/timesrb.jhf fixed/timesrb.jhf
removing unwanted linebreak and CR in orig/timesrb.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/timesrb.jhf","gen_c_src/timesrb.h")'
./fix_linebreak.sh orig/gothgrt.jhf fixed/gothgrt.jhf
removing unwanted linebreak and CR in orig/gothgrt.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/gothgrt.jhf","gen_c_src/gothgrt.h")'
./fix_linebreak.sh orig/greekc.jhf fixed/greekc.jhf
removing unwanted linebreak and CR in orig/greekc.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/greekc.jhf","gen_c_src/greekc.h")'
./fix_linebreak.sh orig/mathupp.jhf fixed/mathupp.jhf
removing unwanted linebreak and CR in orig/mathupp.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/mathupp.jhf","gen_c_src/mathupp.h")'
./fix_linebreak.sh orig/rowmans.jhf fixed/rowmans.jhf
removing unwanted linebreak and CR in orig/rowmans.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/rowmans.jhf","gen_c_src/rowmans.h")'
./fix_linebreak.sh orig/timesi.jhf fixed/timesi.jhf
removing unwanted linebreak and CR in orig/timesi.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/timesi.jhf","gen_c_src/timesi.h")'
./fix_linebreak.sh orig/cursive.jhf fixed/cursive.jhf
removing unwanted linebreak and CR in orig/cursive.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/cursive.jhf","gen_c_src/cursive.h")'
./fix_linebreak.sh orig/gothitt.jhf fixed/gothitt.jhf
removing unwanted linebreak and CR in orig/gothitt.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/gothitt.jhf","gen_c_src/gothitt.h")'
./fix_linebreak.sh orig/rowmant.jhf fixed/rowmant.jhf
removing unwanted linebreak and CR in orig/rowmant.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/rowmant.jhf","gen_c_src/rowmant.h")'
./fix_linebreak.sh orig/gothicita.jhf fixed/gothicita.jhf
removing unwanted linebreak and CR in orig/gothicita.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/gothicita.jhf","gen_c_src/gothicita.h")'
./fix_linebreak.sh orig/scriptc.jhf fixed/scriptc.jhf
removing unwanted linebreak and CR in orig/scriptc.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/scriptc.jhf","gen_c_src/scriptc.h")'
./fix_linebreak.sh orig/gothicger.jhf fixed/gothicger.jhf
removing unwanted linebreak and CR in orig/gothicger.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/gothicger.jhf","gen_c_src/gothicger.h")'
./fix_linebreak.sh orig/futural.jhf fixed/futural.jhf
removing unwanted linebreak and CR in orig/futural.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/futural.jhf","gen_c_src/futural.h")'
./fix_linebreak.sh orig/rowmand.jhf fixed/rowmand.jhf
removing unwanted linebreak and CR in orig/rowmand.jhf
octave -q --eval 'addpath("../src/"); gen_C_code("fixed/rowmand.jhf","gen_c_src/rowmand.h")'
rm fixed/futural.jhf fixed/japanese.jhf fixed/cyrilc_1.jhf fixed/timesr.jhf fixed/scripts.jhf fixed/meteorology.jhf fixed/astrology.jhf fixed/greek.jhf fixed/rowmans.jhf fixed/cursive.jhf fixed/mathlow.jhf fixed/music.jhf fixed/greekc.jhf fixed/timesi.jhf fixed/gothgrt.jhf fixed/gothicita.jhf fixed/timesrb.jhf fixed/greeks.jhf fixed/timesib.jhf fixed/gothicger.jhf fixed/rowmant.jhf fixed/gothitt.jhf fixed/symbolic.jhf fixed/gothgbt.jhf fixed/cyrillic.jhf fixed/mathupp.jhf fixed/markers.jhf fixed/scriptc.jhf fixed/futuram.jhf fixed/gothiceng.jhf fixed/rowmand.jhf fixed/timesg.jhf
make[1]: Leaving directory '/home/pi/dev/hf2gcode/hershey_fonts'
gcc -Wall -Wextra -O0 -ggdb hf2gcode.c libhf2gcode.c -o hf2gcode
libhf2gcode.c: In function ‘get_gcode_line’:
libhf2gcode.c:406:12: warning: comparison is always false due to limited range of data type [-Wtype-limits]
       if (x==-50 && y==0) /*Pen-Up*/
            ^~
./hf2gcode -i demo.txt -o demo_hello_world.ngc -y 23
./hf2gcode -m -s 0.3 -p 2 -f 800 -i demo.txt -o demo_hello_world_needler.ngc -y 23
./hf2gcode -i demo.txt --font=scripts -o demo_hello_world_scripts.ngc -y 23

but when I open the ngc file, it doesn't have the letters I am expecting. I am wondering what I missed. Thank you.

This is what I get:
[image]

[demo_hello_world_scripts.txt](https://github.com/Andy1978/hf2gcode/files/3768733/demo_hello_world_scripts.txt)

Andy1978, 2019-10-24:
Strange. Have you renamed the generated demo_hello_world_scripts.ngc to demo_hello_world_scripts.txt? What architecture is this? Please add the output of "uname -a"

seesunmoon, 2019-10-24:
Thanks for getting back to me.

I renamed gnc to txt, in order to upload the file here.

It is a Raspberry Pi
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Thanks

seesunmoon, 2019-10-24:
Here is what I get from a simple command:

> pi@octopi:~/dev/hf2gcode/src $ ./hf2gcode h
( generated with hf2gcode 0.2.1 )
G21 ( base unit mm )
G90 ( absolute distance mode )
G94 ( Feed Rate Mode: Units per minute Mode )
G17 ( X-Y plane )
M3 S10000
( text="h", font="rowmans" )
( scale=0.500000, feed=200.000000, precision=3 )
F200.000
G0 Z1.000 ( Pen-Up at start)
;h = MFM[ RMQPNRMUMWNXQX[
G0 X2.000 Y-122.000
G1 Z-1.000 ( Pen-Down )
G1 Y-4.500
G1 X-20.500 Y0.000
G1 X2.000 Y-127.500
G1 X3.500 Y-126.000
G1 X-123.500 Y-125.500
G1 X-122.000
G1 X-121.000 Y-126.000
G1 X-120.500 Y-127.500
G1 Y-4.500
G0 Z1.000 ( Pen-Up, EOG )
M5 (stop the spindle)
M30 (Program stop, rewind to beginning of program)

Andy1978, 2019-10-24:
Aaaah, this is armhf where a char is unsigned per default... Please try to change libhf2gcode.c line 404 and 405 from char -> "signed char"

Andy1978, 2019-10-24:
I've made a cset: 9e5a26c05dcc056a5ae6105491b187a7a66402be
Perhaps I have access to a Rpi the next few days and try it out

seesunmoon, 2019-10-24:
It works now. Thank you very much.

I am working on a project, and this tool helped a lot.
I will show you when I am done.

Thanks again.

Andy1978, 2019-10-27:
So the cset was sufficient? And yes, please add some pictures of the resulting project.

sunnyspeedstudio, 2019-10-27:
Totally, it works very well for my need.

https://github.com/sunnyspeedstudio/3d_typewriter

Here is the project, I am still working on it. Should finish it in the next couple days.

Thanks

sunnyspeedstudio, 2019-10-27:
BTW, it is still me, with a different account.

sunnyspeedstudio, 2019-10-27:
added a photo at https://github.com/sunnyspeedstudio/3d-typewriter, if you are interested to see

sunnyspeedstudio, 2019-10-28:
Hi there,

The code is up, and the project is done: https://github.com/sunnyspeedstudio/3d-typewriter

Thanks again for the help.

seesunmoon, 2019-10-28:
Solved

closed 2019-10-28 by seesunmoon