Andy1978/hf2gcode #10: Thank you for this project

pepijndevos, 2023-05-20:
[image]

Andy1978, 2023-11-12:
Hi pepijndevos, very cool.
Sorry for the late reply, looks like I have missed the notification. Do you have a blog or project page for this? I would add a "usage examples" chapter with links to the README and include this

-- Andy

pepijndevos, 2023-11-12:
I'm still working on a project page. All I have now is some twitter threads and an undocumented repository with code. And tbh in the months that past I've switched to vpype for my plotting needs, to which I've contributed hyphenation. Their font support isn't as good though.

Andy1978, 2023-11-13:
Ah, I never heard of vpype before. The initial aim for hf2gcode was to run on a small microcontroller (8-bit RISC) together with grbl (the base of the later well known marlin 3D printer firmware).