Andy1978/hf2gcode #11: fastest font with minimum number of gcode instructions?

monperrus, 2023-11-11:
Hi @Andy1978 

Thanks for the great tool.

I'm looking for the font that is fastest to penplot.

Such a font:
* would have a minimum number of gcode instructions
* would heavily use G02/G03 instructions

Have you ever heard about this? What do you think?

Andy1978, 2023-11-12:
Hi @monperrus

I think if you want to calculate the "fastest" way to plot a font you would have to consider the kinematics of your hardware: max jerk, acceleration and speed (for pens often the ink flow is a problem).

I doubt that a file with minimum of gcode instructions automatically will give you the best throughput.

When I searched for a way to engrave metal using a pneumatic needle pin years ago, the best thing I found was the original hershey fonts, all made of straight lines. Never stumbled over a font using arcs in this time but maybe someone made one?

-- Andy

monperrus, 2023-11-13:
thanks for the feedback Andy!